lumokitho/esp32-keyble #1: Is this working?

corgan2222, 2021-06-14:
I tried your code some month ago, but could not get it working. 
What's about user registration? Can i use the key from my raspi-pi?

corgan2222, 2021-06-14:
Awesome, after hours of trying getting "timeouts" it works now! 

The problem was, that the esp don't work, if connected to the PC. Maybe it gets not enough power.
After connecting to a normal power plug, it works.
Don't know if the mac is case-sensitive, but a changed that at the same time, so maybe this can be a pitfall too.

I really hope, that this solution works better/more reliable as the nodejs solution.

Thanks for your work! 

ps: is there a way to password protect the config website?

lumokitho, 2021-06-25:
Hallo Stefan!

Wie ich sehe kommst Du aus Deutschland, daher erlaube ich mir Dir nicht auf Englisch zu antworten.

Danke für Deine Rückmeldung. Die Änderungen habe ich ja erst vor knapp 2 Wochen eingestellt, Du kannst meine Version also nicht schon vor Monaten ausprobiert haben.

Ich hatte genau das gleiche Problem mit der unzureichenden Stromversorgung und habe das ebenfalls mit einer externen Versorgung gelöst.

Zur Zeit gibt es noch so gut wie keine Fehlerbehandlung, alles ist noch recht grob, teilweise redundant und . Sobald es meine Zeit zulässt, baue ich mal entsprechende Prüfungen ein.

Die Übersichtsseite könnte man recht simpel über htaccess absichern (was aber eher unschön bzgl. fester Werte ist), AutoConnect sieht ebenfalls eine Kennwortabfrage vor, die sich auch recht einfach aktivieren lassen sollte, da muss ich mich aber einlesen.

Auf meiner Liste steht aber an erster Stelle das Anlegen von weiteren Benutzern. Dann geht es mir um die Koexistenz von WiFi/BLE und diese stabil zum Laufen zu bringen. Das scheint daran zu scheitern, das der Austausch der Nonce bei sendMessage noch keine Prüfung enthält. Eine erste Version habe ich bereits im Testlauf, hier kommt aber noch zu unregelmäßigen reboots.

Wenn das soweit läuft, gehe ich mal ans Aufräumen und die Fehlerbehandlung. Es gibt noch einige Ideen zur Erweiterung, z.B. Reboot via MQTT, Heartbeat und NTP, zyklisches Abfragen des Schlosses über den ESP anstelle eines externen Triggers, Funktion des Buttons am ESP konfigurierbar machen,  etc.

Die nodejs Lösung von oyooyo läuft, mit kleinen Anpassungen, bei mir seit über einem Jahr sehr stabil. Damit spreche ich 5 Schlösser gleichzeitig an, die permanent verbunden sind. Läuft auf einem Pi Zero mit USB Hub Hat mit Ethernet und 4 einfachen USB Dongeln. WiFi nutze ich am Pi nicht. Vorteil der permanenten Verbindung ist, das es keine spürbare Verzögerung gibt.

Schönen Gruß
Thomas

corgan2222, 2021-06-25:
Moin Thomas, danke für Deine ausführliche Antwort!
Ich hatte mir meine Fragen einfach selber beantwortet und die meisten Dinge die ich brauchte eingebaut:
https://github.com/corgan2222/esp32-keyble

 new features: watchdog, status led, webserver auth
- watchdog set WDT_TIMEOUT (sec) after this time the esp will reset if something hangs up
- status LED on GPIO32: constant light=wifi conected, slow blink: mqtt conected and waiting, fast blink=conected to keyble
- webserver auth, set WEBSERVER_USER and WEBSERVER_PW. Default admin:admin

Aber genau wie bei der Raspi4 Lösung mit NodeJS und mit einem PiZero läuft das im Büro während der Testphase super.
Sobald das Schloss aber an der Tür ist und ich das Live in Betrieb nehme, gibt es immer wieder Aussetzer.

z.b. Befahl öffnen wird gesendet. Der ESP gibt ein "working" zurück. Springt zurück auf waiting, aber es passiert nix. Nach dem dritten Klicken reagiert das Schloss dann mal.

Schlimmer ist aber, dass ab und zu nach einem reboot, bzw. Strom aus/an, der ESP die WiFi Settings nicht mehr findet und dann sein eigenes Setup Wlan macht. 

Ich bin langsam am Zweifeln, ob ich das Schloss wirklich jemals vernünftig und stabil zu Laufen bekomme. Egal ob RASPI4, Zero oder ESP, irgendwie scheint das Schloss die Verbindung dicht zu machen. Ich hatte irgendwo so einen verrückten Hack gesehen, dass Leute die Open/Close Buttons am Schloss direkt mit nem ESP schalten. Ich bin auch kurz davor. :)

Auf welche Timeouts hast du die Verbindung gesetzt, wenn du sagst, dass du immer verbunden bist? Wie lange machen die Batterien das mit?

lumokitho, 2021-06-25:
Servus Stefan!

Wie sieht es mit der BLE Signalstärke aus? Bist Du ausschließlich mit dem ESP/Pi mit dem Schloss verbunden? Falls Du gleichzeitig auch noch die original App verwendest, kann es sein, dass diese im Hintergund läuft und weiterhin sich mit dem Schloss verbindet. Das führt zu instabilem Verhalten. Nutzt Du am Pi das interne BLE Modul oder ein Dongel? Die interne Verbindung funktioniert aus meiner Erfahrung unzuverlässig, wenn das Schloss zu weit weg ist.

Deine Erweiterungen gucke ich mir bei nächster Gelegenheit mal an.

Schönen Gruß
Thomas

corgan2222, 2021-06-25:
Der ESP hängt ca. 20cm neben dem Schloss. Das war nie das Problem. Es war eher so, dass die Verbindung stand, aber der Befehl einfach nicht ausgeführt wurde. In den Amazon Kommentaren berichteten auch andere User von dem Problem. Ansonsten gibts da kein Gerät was aufs Schloss zugreifen kann. Das war aber auch mein erster Gedanke.

Ja, sowohl beim Raspi4, als auch beim raspi zero hatte ich das interne BT verwendet. Wobei beim pi4 wifi deaktiviert war. 
Welchen BT Dongle nutzt du denn am zero?

Der Code ist jetzt nicht wild. Quasi copy&paste aus anderen Tutorials. Ich wollte ne LED Anzeige, so dass ich sehen kann, in welchem Zustand das Schloss gerade ist. Also WiFi oder BT. 

vg Stefan

lumokitho, 2021-06-25:
Hi!

Welche Firmwareversionen hat das Schloss bei Dir?

Bzgl. Dongle; CSR 4.0 gibt es bei Amazon für ~3€. Aber Vorsicht, die haben sehr häufig identische MAC Adressen. Google wirft aber sofort mehrere Treffe bei "CSR MAC changer" raus, damit kann man das recht einfach ändern. Praktisch, falls man mehrer verwendet, weil man dann die MAC selbst bestimmen kann.

Gruß
Thomas

Schönen Gruß
Thomas

lumokitho, 2021-06-25:
Hi Stefan,

ich hatte ganz übersehen, dass Du auch nach dem Timeout gerfragt hast, das steht bei mir auf 15 Sekunden in der Pi Version. Je nach Batteriehersteller halten die ~2 Monate bei Dauerverbindung.

Gruß Thomas

corgan2222, 2021-06-25:
ok danke für die Infos! vg

pooyashahidi, 2021-09-22:
Hi @corgan2222,

I had wifi problems, like the device creating it's own network instead of trying to connect to the already saved network. Maybe it's similar to the one you mentioned above?
After adding `config.autoReconnect = true;` I have never experienced any such problems.

corgan2222, 2021-09-23:
Nice tip! Where did you put this line in?

pooyashahidi, 2021-09-23:
Sorry I forgot to mention that. In the `Setup`.
like [other autoconnect configs](https://github.com/lumokitho/esp32-keyble/blob/b076c6903956f31313abdad882eac817bc8e1961/src/main.cpp#L513):

corgan2222, 2021-09-23:
Ah ok, I will try this! Sometimes this thing "forgot" the Wi-Fi setting, or something else. I noticed that sometimes it hangs in the default state with 192.168.4.1 open for config. 
Hope that this fix this. If not, I will try to change the init process to https://github.com/ESPresense/ESP-WiFiSettings which is really nice and much cleaner and easier to use.

pooyashahidi, 2021-09-25:
That is probably a better solution. Unfortunately mine just started acting up -_-
It was working for a couple of days but it is doing it again  After pressing the reset button it connects to the wifi again but it is not reliable... sorry for giving false hope.
[image]

pooyashahidi, 2021-09-27:
update: even with autoReconnect and reconnectInterval set, I still have connection issues. I think I wasted way too much time on this :( So as a workaround I just made an automation that restarts the device when it is not connected to wifi for more than 1.5 minutes.

corgan2222, 2021-09-27:
How have you done this automation? In the ESP or HA?

corgan2222, 2021-09-27:
I have the problem, that once in 3-4 days the ESP hangs on status working. I have to hard reset it after this. But tilll 2 Days it works fine.

[image]

pooyashahidi, 2021-09-28:
well I'm not really proud of the way I did it, but I basically cut its power for a couple of seconds. lol
mine does it  randomly sometimes a couple of times per day or sometimes not at all.

corgan2222, 2021-09-28:
hehe, ok, that's really a funky workaround, but at the end of the day this thing has to be working. :)
Do you use my fork, or the original?
My solution is even more overkill, because  atm, I'm using an extra esp8266-wifi switch to reset the device, because if the ESP hangs in the "working" state, only a hard reset helps. 

But I have some ideas which hopefully fix all this crapy workarounds.

I like your button style! Might you share your loveace yml?

pooyashahidi, 2021-09-28:
I mean sure it's not Ideal but it was the quickest thing I could put together and had the hardware laying around :D
That's great to hear! I look forward to it. 

Yup I'm using yours, commented out the led part as my board does not have an LED (cheap basterds!) and added a relay to buzz the intercom. The only suggestion that I have is to change the watchdog timeout ( I don't really know what problem it solves) but the short timeout prevented OTAs for me.

About the lovelace card, if you mean the appearance, it's based on a great work by [matt8707](https://github.com/matt8707/hass-config/blob/5eea9f20921ef36a4c415c36b45981c13b79c941/button_card_templates.yaml#L1049).

corgan2222, 2021-09-28:
Ok, I will test and come back here, if there are any news.
Thanks for the link. :)

maijo366, 2022-03-18:
Muss ich das Schloss per bluetooth noch mit dem ESP pairen od. wird die Verbindung automatisch erkannt. 

Was muss ich in Homeassistant (config.yaml od. sensor.yaml)  eintragen, damit die Sensoren bzw. Entitäten erkannt werden. MQTT hat Daten erhalten bzw. den Client erkannt.

per IP des ESP sehe ich die config und verbindungsdaten des ESP im Netzwerk. 

Vielen Dank