lumokitho/esp32-keyble #4: Create/use existing user

tomcat-xx, 2021-08-24:
Hi, I thought I could quickly use one of my existing ESP32 and connect to the new lock. To late I saw "TODO: register user feature". Do you know if I can use existing user (created with the original app on one mobile)?

tomcat-xx, 2021-08-25:
Maybe I can answer my own question: Seems that it is possible, as user can not be created in the ESP an anyway has to be from other device, so MAC can not be restricted. Only "seems" because I got a connection 2-3 times but now not anymore, anyway issue should not be the user anymore. But I'm wondering if pairing mode has to be on (one of my successful tries it was) and what kind of user is needed (has it to be administrator)?

JBrumby, 2022-03-14:
No, the original android app doesnt work. You habe to use the commandline tools to create a new user with you master key. See issue 10 and 11 how I got it working.
Regards