lumokitho/esp32-keyble #6: Lockstatus -1 - timeout

pooyashahidi, 2021-09-16:
I connected the esp32 to mqtt and wifi no problem but the lock always gives a timeout.
Is there anything needed beside the steps in the readme?
I don't know if the key is wrong or what. I tried using the whole key card as "KeyBLE User Key", I tried using only the 32 characters after the "K" also tried it in lower case and upper case and nothing works. 
The Mac address is correct as I get a value for RRSI.
And for the user id I tried 1 and 2.

The lock is working using the app. 
Am I missing something?
thank you :)

corgan2222, 2021-09-16:
I had similar problems. 
First, you have to add a new user with the keyble commands from another device, like a Raspberry Pi.
https://github.com/oyooyo/keyble 

After that, you can use the credentials from that user in the esp32-keyble. Make sure, that no other device/app is trying to communicate with the lock.

pooyashahidi, 2021-09-16:
@corgan2222 thank you!
That cleared some of my confusions. I created the user with a Raspberry Pi and it in fact works with the cli. But the same key and id does not work with esp32.
I still get the same timeout error :(
I even put the esp and lock in a separate room far away from all my bluetooth devices and still the same result

pooyashahidi, 2021-09-16:
Okay after factory reseting the lock a couple of times and trying again and again it works now...  I have no clue why or why it did not work before :/

closed 2021-09-16 by pooyashahidi

corgan2222, 2021-09-16:
> 
> 
> Okay after factory reseting the lock a couple of times and trying again and again it works now... I have no clue why or why it did not work before :/

ok great!