lumokitho/esp32-keyble #10: Welche Verbindungseinstellungen? MQTT and KeyBLE credentials??

maijo366, 2022-03-08:
[image]

KeyBLE UserID?
KeyBLE MAC?
MQTT Topic?

Wie kann ich eine Verbindung (Bluetooth) zum Eqiva Q3 Smart Türschlossantrieb herstellen?

MfG
Bitte um Hilfe

JBrumby, 2022-03-14:
Das mqtt-topic ist beliebig. Ich habe dafür den Namen meines Schlosses genommen. z.B. "keyble". auf dem MQTT-Server erhältst du dann Daten wie:

keyble/battery true
keyble/rssi -61
keyble/command 3
keyble/task working
keyble/state locked

Du must wirklich einen User auf dem Schloss anlegen, dazu verwendest du folgendes:

Auf dem github: https://github.com/oyooyo/keyble findest du eine Anleigung wie du die notwendigen Programme installierst. Du benötigst einen Bluetoothdongle oder einen PI mit bluetooth (4, 400)

keyble-registeruser [-h] [--user_name USER_NAME]
                           [--qr_code_data QR_CODE_DATA]

dafür musst du mit einem QR-Code-Reader deinen Masterschlüssel von der mitgelieferten Karte auslesen, den du mit:
   --qr_code_data an das Programm gibst.
du denkst dir einen neuen Namen aus z.B. "John"  und übergibst ihn ebenfalls auf der command-line mit --user_name

Wenn du das Programm startest, fordert es dich auf den Öffnungs-Button am Schloss solange zu drücken, bis die LED gelb blinkt.

Es wird dann im Schloss ein neuer User angelegt mit einer Nummer als ID und einem key für diesen User. Du bekommst ebenfalls die mac-adresse angezeigt.
Schreib die Daten genaus so in die Felder deiner Anfrage und es sollte gehen.

- alles Kleinschreibung. 
- Mac mit Doppelpunkten getrennt, 

Viel Erfolg

maijo366, 2022-03-16:
Hallo, 
vielen Dank für die Nachricht.
Ich habe den User angelegt und bekommen nun folgende Rückmeldung: KeyBLE last battery state: 1
KeyBLE last command received:
KeyBLE last rssi: -85
KeyBLE last state: locked
KeyBLE last task: waiting

Was muss ich in Homeassistant (config.yaml od. sensor.yaml) noch eintragen, damit die Sensoren bzw. Entitäten erkannt werden. MQTT hat Daten erhalten bzw. den Client erkannt.

Wie steuerst du @JBrumby das Schloss?

artur90226, 2022-03-25:
@maijo366 
hi, bist du weitergekommen?
meine integration im HA erkennt den schloss nicht mal

artur90226, 2022-03-25:
@JBrumby 
genau so gemacht wie du es beschreibst, aber bekomme keine infos von dem Schloss