lumokitho/esp32-keyble #12: Befehle per MQTT? Einbindung in HA?

maijo366, 2022-03-24:
Hallo,
Ich habe leider als Anfänger etwas Probleme damit. Es wurde von mir ein User angelegt und ich kann den ESP32 auch erreichen, bzw. bekomme ich auch die Statusmeldungen per IP des ESP. Leider kann ich weder mit der Togglefunktion am Button des ESP noch per Cmnd Befehle ausführen um das Schloss anzusteuern. Es tut sich einfach nichts. 
 
Per MQTT.fx sehe ich die Daten. Wenn ich den Toggle Button am ESP betätige steht kurz working. Am Schloss tut sich nichts. 
Wie lauten die genauen Befehle zum Ansteuern (Publish) des Schlosses, bzw. wie kann ich es Steuern.
Ich kann mit der Anleitung von [oyooyo](https://github.com/oyooyo)/[keyble](https://github.com/oyooyo/keyble) das Schloss steuern.

Was muss ich in HA in den config.Yaml und Sensor.yaml eintragen?
Ich möchte das Schloss in HA einbinden.

Den ESP32 habe ich mit PlatformiO geflasht.

Sehr viele Fragen und ich bin mittlerweile schon am verzweifeln.🤔

Bitte um Hilfe

artur90226, 2022-03-26:
Hi, bist du weitergekommen? Ich stehe auch auf dem Schlauch.

maijo366, 2022-03-28:
Ja endlich, es funktioniert.

> Hi, bist du weitergekommen? Ich stehe auch auf dem Schlauch.

artur90226, 2022-03-28:
Kannst du mir da weiterhelfen? Ich bekomme keinen Status in mqtt Server

maijo366, 2022-03-29:
> Kannst du mir da weiterhelfen? Ich bekomme keinen Status in mqtt Server

Gerne, wenn ichs kann ;-) deine emailadr.?

artur90226, 2022-03-29:
> > Kannst du mir da weiterhelfen? Ich bekomme keinen Status in mqtt Server
> 
> Gerne, wenn ichs kann ;-) deine emailadr.?

artur.kaul@gmail.com
Benutzt du dieses Projekt oder von tc-maxx?

sveba, 2022-04-11:
@maijo366 oder @artur90226  paste dein yaml einfach hier. Ich glaube da wären mehrere daran interessiert

jelflein, 2022-05-09:
lock:
  - platform: mqtt
    name: Frontdoor
    state_topic: "lock/state"
    command_topic: "lock/command"
    payload_lock: "3"
    payload_unlock: "2"
    payload_open: "4"
    state_locked: "locked"
    state_unlocked: "unlocked"
    optimistic: false
    qos: 1

DaHype, 2023-02-25:
@jelflein 
Stehe auch vor dem Problem. Kann Lock und Unlock per Button auslösen am ESP32 und sehe auch im WebInterface von lumokitho den Status. Im Home Assistant bekomme ich die Daten aber nicht eingelesen. Im MQTT Broker erscheinen sie nicht, und wenn ich den lock aus deinem Post in die YAML einfüge ist die YAML danach Fehlerhaft.

Wäre für Hilfe sehr dankbar!

DANKE DIR!

DaHype, 2023-02-26:
Im MQTT Explorer unter Windows wird mit das Schloss angezeigt sobald ich mich mit dem HAS MQTT Broker verbinde.
Ebenfalls lässt sich das Schloss im Explorer triggern und ändert auch den Status korrekt.

Im HAS ist aber von der Entität keine Spur. Jemand eine Idee?