lumokitho/esp32-keyble #16: Need to reassign wifi credentials on reset

mourraille, 2022-05-22:
Anyway I can make the esp32 automatically reconnect to my preferred SSID on reboot? Whenever theres a power cut or something like that it requests me to select a wifi network every time. Cant it reconnect on its own ?

tomcat-xx, 2022-05-30:
For me even after re-flashing (without erasing flash) the wifi settings are still there.

As I now after longer break finally made it to add users with keyble installed my Raspberry Zero W I also flashed the esp32 again. Did not remember which version was on and wanted to be sure. This time I did it without erasing the flash (requested in the step by step instruction). All settings were still there (wifi and also mqtt),

mourraille, 2022-05-30:
I made it work by hardcoding credentials under `Portal.begin(SSID,PASS)` 🤷🏻‍♂️

marcelk1101, 2022-12-18:
> I made it work by hardcoding credentials under `Portal.begin(SSID,PASS)` 🤷🏻‍♂️

Can you please exactly what to do, in which lines!? My ESP32 also lost the connection after some time? What changes you made in code? Thank you.

marcelk1101, 2023-03-17:
> > > I made it work by hardcoding credentials under `Portal.begin(SSID,PASS)` 🤷🏻‍♂️
> > 
> > 
> > Can you please exactly what to do, in which lines!? My ESP32 also lost the connection after some time? What changes you made in code? Thank you.
> 
> Did you found an solution? Same Problem... Thanks

No, need some help from @mourraille

DaHype, 2023-03-20:
I hope @mourraille can explain it. :-(
It workls nearly perfect, but each day i have to reconnect again manual to my network :-( :-(

DaHype, 2023-03-26:
I contaced @mourraille with email and Insta but he don´t write back.
I really hope someone can help here with that. 
It's so awkward having to dial in every day. Thank you!

DaHype, 2023-03-26:
// ---[SetupWiFi]---------------------------------------------------------------
void SetupWifi()
{
  digitalWrite(LED_GPIO,LOW);

  if (Portal.begin())
  {
   if (WiFi.status() == WL_CONNECTED)
   {
     Serial.println("# WIFI: connected to SSiD: " + WiFi.SSID());
     digitalWrite(LED_GPIO,HIGH);
   } 
   int maxWait=100;
   while (WiFi.status() != WL_CONNECTED) {
    Serial.println("# WIFI: checking SSiD: " + WiFi.SSID());
    delay(500);
    
    if (maxWait <= 0)
        ESP.restart();
      maxWait--;
  }
  Serial.println("# WIFI: connected!");
  Serial.println("# WIFI: signalquality: " + String(GetWifiSignalQuality()) + "%");
  Serial.println("# WiFi connected to IP: " + WiFi.localIP().toString());
  digitalWrite(LED_GPIO,HIGH);
  }

So it means i have to insert here SSID and PW from my Wifi?

 if (Portal.begin(MY SSID) MY PASSWORD)

??

DaHype, 2023-04-07:
It seems no one will or can help with that :-(

DaHype, 2023-05-07:
Please need help for Hardcoding the Wifi,  each day i have the Problem that the ESP32 lose the connection to the Wifi and than shown the Wifi keyble and i have to set up again. Than it works some hours only! 

HELP!

tomcat-xx, 2023-05-14:
What I did is:
if (Portal.begin("SSID","key"))

For me that helps when ESP32 lost power or I restart it.

But ... I'm still having issues when WIFI was off (AP restart, AP without power but not ESP32). Then connection get lost and I have to manually restart keyble.

tomcat-xx, 2023-05-14:
Ah, sorry, I thought it is about how, now where. Because the line is in your comment from March 26 already. So in main.cpp (in my case renamed to main.ino as I used Arduino IDE to flash).

For me it is in line 531, but I'm using corgan2222/esp32-keyble. Here it seems to be line 474