lumokitho/esp32-keyble #17: Esp32 unavailable after some time

artur90226, 2022-05-25:
Hi,
I have the problem that it is not available after some time. Only reset button helps. Then it's immediately online.

tomcat-xx, 2022-06-01:
Could it be that wifi signal is lost or was before? Do you then see SSID KeyBLEBridge again? That's something that I saw, but just for one time now.

artur90226, 2022-07-13:
I can't see the esp under my router, and for example now it's not visible for 7 days. It needs a manual restart, then it goes again. So far, I have not been able to determine why this is the case.

artur90226, 2022-07-22:
I found out the following. I could connect to the wifi of esp. Then reconfigured aP and it worked. Only it is strange that it does not automatically connect to a wifi.

DaHype, 2023-02-28:
Same Here.. I was happy all worked fine about 14 hours. Than No Response and i See keyble AP. I Had to reconfig my AP and all fine again... My WiFi 100% and No Internet Problems....