lumokitho/esp32-keyble #19: problem compiling

Urdi69, 2022-10-19:
Hello I have big problems when compiling the code

` *  Ejecutando tarea: platformio run 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32 BLE Arduino @ 1.0.1
|-- PubSubClient @ 2.8.0
|-- AutoConnect @ 1.2.3
|   |-- PageBuilder @ 1.5.0
|   |   |-- FS @ 2.0.0
|   |   |-- LittleFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- SPIFFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WebServer @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.18.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- EEPROM @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- Ticker @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- HTTPUpdate @ 2.0.0
|   |   |-- HTTPClient @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |   |-- WiFi @ 2.0.0
|   |   |-- Update @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- eQ3
|   |-- ESP32 BLE Arduino @ 1.0.1
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- WebServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Building .pio/build/esp32dev/bootloader.bin
esptool.py v4.2.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generating partitions .pio/build/esp32dev/partitions.bin
WARNING: Partition spiffs not aligned to 0x1000.This is deprecated and will be considered an error in the future release.
Compiling .pio/build/esp32dev/lib569/ESP32 BLE Arduino/BLE2902.cpp.o
In file included from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLE2902.h:13,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLE2902.cpp:15:
.pio/libdeps/esp32dev/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio/build/esp32dev/lib569/ESP32 BLE Arduino/BLE2902.cpp.o] Error 1
In file included from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLERemoteDescriptor.h:18,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLERemoteCharacteristic.h:18,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLERemoteService.h:16,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEClient.h:19,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEScan.h:17,
                 from lib/eQ3/eQ3.h:4,
                 from src/main.cpp:6:
.pio/libdeps/esp32dev/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1
================================ [FAILED] Took 16.79 seconds ================================

 *  El proceso del terminal "platformio 'run'" finalizó con el código de salida 1. 
 *  Las tareas reutilizarán el terminal, presione cualquier tecla para cerrarlo. 
`

leolurch, 2022-11-09:
Same problem for me

geni0602, 2022-11-09:
In your platformio.ini just try an older ESP32 BLE Adruino version under lib_debs
for example:
lib_deps =
    ESP32 BLE Arduino @ 1.0.1

and try to compile again.

markusmauch, 2022-12-18:
> In your platformio.ini just try an older ESP32 BLE Adruino version under lib_debs for example: lib_deps = ESP32 BLE Arduino @ 1.0.1
> 
> and try to compile again.

Arduino @ 1.0.1 is initially in there. The project still does not compile:
```
.pio/libdeps/esp32dev/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
```

Sami6053, 2022-12-29:
Same problem for me

MariusSchiffer, 2023-02-24:
Remove the library entirely, it was included in the core arduino library