lumokitho/esp32-keyble #20: Problem bei der Installation PlatformIO

maijo366, 2022-10-22:
Derzeit lässt sich der Code nicht mehr installieren. Fehlermeldungen werden angezeigt.
Hat sich in VSCode/Platformio etwas geändert?
Board Olimex BLE ESP32
Bitte um Hilfe

leolurch, 2022-12-08:
Habe das gleiche Problem :/

DaHype, 2023-02-22:
Konnte jemand das Problem schon beheben?

Erhalte ständig : Error 1, Cant Read direct.... Spiffs.bin

DaHype, 2023-02-24:
> mbedtls/aes.h

Vielen Dank. Bin nun schon einen Schritt weiter. Zum Schluss erscheint jetzt das ich über das mermory maximum bin.
Wie lässt sich das noch beheben?

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (2026797 bytes) is greater than maximum allowed (1966080 bytes)
RAM:   [==        ]  19.7% (used 64680 bytes from 327680 bytes)
Flash: [===*** [checkprogsize] Explicit exit, status 1
=======]  103.1% (used 2026797 bytes from 1966080 bytes)

Vielen DANK!

DaHype, 2023-02-24:
@MariusSchiffer 

Zufällig eine Idee was ich noch falsch gemacht hätte wenn kein AP erscheint?
Vielen Dank

Build erfolgreich auf den ESP32 geflashed, aber es erscheint kein AP, 
daher kein Wifi das lautet " KeyBLEBridge "

Wo kann hier das Problem liegen?

Wrote 2032576 bytes (1252792 compressed) at 0x00010000 in 30.0 seconds 
(effective 542.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
======================================================================== 
=================  [SUCCESS]

Danke für Hilfe