lumokitho/esp32-keyble #21: WiFi and BLE at the same time: with use of NimBLE

KlausMu, 2024-02-19:
Hi, is this repo still maintained?

I've been using the eQ3 lib for some years. It always was annoying that you have to disconnect WiFi before using BLE and vice versa. Establishing the connection always takes some time, both for WiFi and BLE. The keyBLE connection always took about 10-15 sec, sometimes even more.

Recently I integrated the use of [NimBLE](https://github.com/h2zero/NimBLE-Arduino) into the eQ3 lib. With that, WiFi and BLE both can stay connected all the time. Works now for some months without any problems.

Since the timeout for the BLE connection is 4 minutes, I do an `keyble->updateInfo();` each 3 min 50 sec. With that, the keyBLE stays super responsive all the time. You cannot recognize any delay.

Interested in a pull request?

KlausMu, 2024-10-07:
What do you mean with "does it also work with ESPHome"? This project does not work directly with ESPHome. I saw that there are some ESPHome projects using the (more or less) same eQ3 library as the one from this repository:
https://github.com/digaus/esphome-components-eqiva
-> uses a changed eQ3-library
https://github.com/mccoy88f/esp32-keyble-ha
-> uses the unchanged eQ3-library
So what do you already have running? What is the obstacle with it? And what do you want to achieve?

wladimir-computin, 2024-10-14:
Hi @KlausMu!

Could you please share your NimBLE version of the eQ3 lib? This would be awesome, it is exactly what I need for my project.

I'm using my very own smarthome ecosystem, written from scratch and started way before HA was a thing.
Long story short, I just need the eQ3 lib from this project since I'd like to implement this lock in it. It heavily relies on the WIFI connection, so dropping WIFI every time to perform BLE stuff would be a major PITA ^^

KlausMu, 2024-10-14:
@wladimir-computin 
Here it is:
https://github.com/KlausMu/esp32-keyble
You have to add
```
lib_deps = 
	h2zero/NimBLE-Arduino@^1.4.2
```
to `platformio.ini`

And this is the pull request:
https://github.com/lumokitho/esp32-keyble/pull/23

wladimir-computin, 2024-10-15:
@KlausMu Thanks a lot! I built upon your work and made some additional changes to the lib. Since we don't need to switch between BT and WiFi, a lot of the complexity could be removed. Even the Semaphores are no longer necessary, at least according to my testing.

Since I modded the library to my needs, it is no longer directly compatible with this project. Anyway, it might still be useful for someone.

I also fixed the user registration / pairing, so no more node.js stuff is needed to get it up and running.

https://github.com/wladimir-computin/eq3-lib-NimBLE